nonNoise/PyMCP2221A #2: Failed to open connection

PaTPouL, 2019-01-11:
HI, 

Thanks for this project, I was looking to do something similar. 

When I try to run your example I got an error during open session 

>   File "/home/ppoulin/workspace/devops_hw/usbsfp/PyMCP2221A/example/HIDAPItest.py", line 18, in <module>
    mcp2221a.open_path(hid.enumerate(0x04D8, 0x00DD)[0]["path"])
  File "hid.pyx", line 72, in hid.device.open_path
OSError: open failed
Process finished with exit code 1

I can see my device using dmesg

> [ 1961.826122] usb 1-6.4: new full-speed USB device number 9 using xhci_hcd
[ 1961.931560] usb 1-6.4: New USB device found, idVendor=04d8, idProduct=00dd
[ 1961.931572] usb 1-6.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1961.931580] usb 1-6.4: Product: MCP2221 USB-I2C/UART Combo
[ 1961.931589] usb 1-6.4: Manufacturer: Microchip Technology Inc.
[ 1961.933515] cdc_acm 1-6.4:1.0: ttyACM0: USB ACM device
[ 1961.941140] hid-generic 0003:04D8:00DD.0006: hiddev0,hidraw3: USB HID v1.11 Device [Microchip Technology Inc. MCP2221 USB-I2C/UART Combo] on usb-0000:00:14.0-6.4/input2

Maybe it's related to my lib import, I was not able to import hidapi
I try pip install hidapi

> ppoulin@ppoulin-ThinkPad-T460:~$ pip install hidapi
Collecting hidapi
Collecting setuptools>=19.0 (from hidapi)
Using cached https://files.pythonhosted.org/packages/37/06/754589caf971b0d2d48f151c2586f62902d93dc908e2fd9b9b9f6aa3c9dd/setuptools-40.6.3-py2.py3-none-any.whl
Installing collected packages: setuptools, hidapi
Successfully installed hidapi-0.7.99.post21 setuptools-40.6.3

But When I try to import in python I got an error

> ppoulin@ppoulin-ThinkPad-T460:~$ python 
Python 2.7.15rc1 (default, Nov 12 2018, 14:31:15) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import hidapi
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named hidapi

That's why I use import hid instead import hidapi.
I try with python 2 and python 3 but I got the same result. 

Maybe I forgot something during your package installation. 

Thanks again for your help, 

PaTPouL

from lsusb -v 

> Bus 001 Device 009: ID 04d8:00dd Microchip Technology, Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         8
  idVendor           0x04d8 Microchip Technology, Inc.
  idProduct          0x00dd 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          107
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      28
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

eN0Rm, 2019-07-02:
I made it work by running as root. 

user@computer:~$ sudo -i 
root@computer:~# cd /PyMCP2221A/
root@computer:/PyMCP2221A# ls
MCP2221_i2cdetect.py  venv
root@computer:/PyMCP2221A# source venv/bin/activate
(venv) root@computer:/PyMCP2221A# python MCP2221_i2cdetect.py 
--------------------------------------------------
MCP2221(A) i2cdetect 
--------------------------------------------------
Reset
   0   1   2   3   4   5   6   7   8   9   A   B   C   D   E   F  
  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --   0F
  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --   1F
  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --   2F
  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --   3F
  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --   4F
  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --   5F
  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --   6F
  --  --  --  --  --  --  --  --  --  --  --  --  --  --  --
(venv) root@computer:/PyMCP2221A#

nonNoise, 2019-07-02:
Thank you!! 
As another way..........
There is also a technique of putting compiled binary data in the folder of PyMCP 2221A, but I can not recommend it.
I'm looking for a good way.

closed 2019-08-27 by nonNoise