nonNoise/PyMCP2221A #3: Docs?

mrpackethead, 2019-06-08:
Are there any documents for using this library?

I've kind of got it working, but not quite

mrpackethead, 2019-06-09:
In the example MCP2221_EEPROM_WriteReadTest.py, there is some code;

for i in range(MAX):
    data=[0]*3
    data[0] = (0xFF00&i)>>8
    data[1] = 0xFF&i
    data[2] = 0xFF&i
    mcp2221.I2C_Write(0x50,data)

I am trying to understand what the signficance of each byte in the data list is, and what that translates to with respect to what is transmitted on the I2c bus.

nonNoise, 2019-06-10:
Sorry for the few comments.

    data=[0]*3         <-  Make 3 arrays
    data[0] = (0xFF00&i)>>8    <- EEPROM Upper address
    data[1] = 0xFF&i                <- EEPROM Lower address
    data[2] = 0x00                    <- data
mcp2221.I2C_Write(0x50,data)

and this is test write.
     data[2] = 0xFF&i   <-  Put the count value into the data.

Thank you.

mrpackethead, 2019-06-12:
Can you tell me how you can set the speed of the I2C bus?

I need a way to hold SDA low for a minimum of 60uS, if i can write 0x00 to 0x00 then that should do the job

nonNoise, 2019-07-02:
PyMCP2221A.py
```
#######################################################################
# I2C Init
#######################################################################
    def I2C_Init(self, speed=100000):  # speed = 100000

        buf = [0x00, 0x10]
        buf = buf + [0 for i in range(65 - len(buf))]
        buf[2 + 1] = 0x00  # Cancel current I2C/SMBus transfer (sub-command)
        buf[3 + 1] = 0x20  # Set I2C/SMBus communication speed (sub-command)
        # The I2C/SMBus system clock divider that will be used to establish the communication speed
       buf[4 + 1] = int((12000000 / speed) - 3)
　　........
```
OK.
Please try this method.
I'm sorry, but I have not confirmed that the function works properly.

I2C_Init(speed=133000)

Please check with an oscilloscope.
Is the speed correct?

closed 2020-02-19 by nonNoise