nonNoise/PyMCP2221A #6: I2C_State_Check return value meaning

bludin, 2019-10-14:
Can you please tell me how to interpret the values that I2C_State_Check() return?
Thanks in advance,
Beat

nonNoise, 2019-10-14:
I2C_State_Check() is Read a 0x10 Register.
Please see the data sheet.

http://ww1.microchip.com/downloads/en/DeviceDoc/20005565B.pdf

Please look at page 23.

3.1 USB HID Commands/Responses
3.1.1 STATUS/SET PARAMETERS
....
....
....

This is the content of I2C_State_Check.

bludin, 2019-10-15:
Yes, it is byte[8] of the Response 1 Structure, described as the Internal I2C state machine state value. But what do the various values (responses I see are 37, 69, 98, etc.) actually mean?

nonNoise, 2019-10-15:
Aaaaa..... I remember !!
When writing this source, I was troubled because I didn't understand the contents of the register in the same way.

I wrote it because the Microchip Linux driver is in C language.

http://ww1.microchip.com/downloads/en/DeviceDoc/mcp2221_0_1.tar.gz

The driver has the following declaration statement.

i2c-mcp2221.c
---------------------------------------------------------------------------------------------
#define RESP_I2C_IDLE			      0x00
#define RESP_I2C_START_TOUT		      0x12
#define RESP_I2C_RSTART_TOUT	      0x17
#define RESP_I2C_WRADDRL_TOUT	      0x23
#define RESP_I2C_WRADDRL_WSEND      0x21
#define RESP_I2C_WRADDRL_NACK	      0x25
#define RESP_I2C_WRDATA_TOUT	      0x44
#define RESP_I2C_RDDATA_TOUT	      0x52
#define RESP_I2C_STOP_TOUT		      0x62
---------------------------------------------------------------------------------------------
Is it right?
I didn't use it as a source because I didn't know it was correct.

(responses I see are 37, 69, 98, etc.) 

37 = 0x25 = RESP_I2C_WRADDRL_NACK ?
69 = 0x45 = ???
98 = 0x62 = RESP_I2C_STOP_TOUT ?

Can someone test it?
If it matches, an error character is displayed.
Thanks.

bludin, 2019-10-16:
Thanks, that's quite helpful already. I cannot verify this, but I can possibly shed some light on the meaning of 37/0x25 and 69/0x45: 
0x25 is the value immediately after the first I2C_Write_No_Stop, before performing a subsequent I2C_Read_Repeated (after which the state is 0x00 again.
For all subsequent operations of this kind, I get 0x45 instead of 0x25.

nonNoise, 2020-02-19:
Wow!! 
When I happened to look at the source of MCP2221A for Android, more detailed status was written.

[image]

69 = 0x45 = I2CM_SM_WRITEDATA_END_NOSTOP