nonNoise/PyMCP2221A #17: IndexError: list index out of range

xsherlockpl, 2021-04-21:
Dear,
I'm trying to run  the example but it fails.

```
sherlock@dt-minion-1:~$ sudo python3 MCP2221_i2cdetect.py
--------------------------------------------------
MCP2221(A) i2cdetect
--------------------------------------------------
Reset
Traceback (most recent call last):
  File "MCP2221_i2cdetect.py", line 13, in <module>
    mcp2221 = PyMCP2221A.PyMCP2221A()
  File "/usr/local/lib/python3.8/dist-packages/PyMCP2221A/PyMCP2221A.py", line 16, in __init__
    self.mcp2221a.open_path(hid.enumerate(VID, PID)[devnum]["path"])
IndexError: list index out of range
```

I can see the device (adafruit board) is connected to the USB.

```
sherlock@dt-minion-1:~$ sudo lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
Bus 001 Device 006: ID 2c7c:0800 Quectel Wireless Solutions Co., Ltd. RM500Q-GL
Bus 001 Device 005: ID 2886:8027 Seeed Technology Co., Ltd. Seeeduino_Cortex_M0+
Bus 001 Device 020: ID 04d8:00dd Microchip Technology, Inc. MCP2221 USB-I2C/UART Combo
Bus 001 Device 004: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 003: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 002: ID 16d0:063d MCS Master Brick
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

is that incompatibility with python3?
What can be causing that?

nonNoise, 2021-04-22:
Thank you very much.
I haven't had a similar issue, but mcp2221.Reset may have a problem.
mcp2221.Reset is a "Hacks" to release by resetting MCP2221A when the connection to I2C fails and it stops at Read.
It's not working properly as a HID driver, so you might want to comment it out.
It will be removed in future improvements.

Also, please tell me the Python version in detail.
I have certainly not tried the new version.
I didn't notice that Python had reached 3.9.
It's changing faster than it was in the Python2 era.

xsherlockpl, 2021-04-22:
I slept with the problem and now it is obvious that the problem is a Reset() function on my setup.  It just crashes the the board.

I use Ubuntu 20.10
Python 3.8.6 (default, Jan 27 2021, 15:42:20)
And that little board from  Adafruit MCP2221A Breakout connected over USB-C
https://www.adafruit.com/product/4471

Without Reset() it works great and I have not seen a crash yet.

closed 2021-04-22 by xsherlockpl